esp-rs/esp-rust-board #5: License

0Grit, 2021-11-20:
Is there a license with which to use this work?

pedrominatel, 2021-11-22:
We need to define the license and create the file.

@jessebraham and @MabezDev can you help on this issue?

MabezDev, 2021-11-22:
Most of our Rust projects are dual licensed MIT or Apache, would that be suitable for this project too?

0Grit, 2021-11-22:
@ChrisGammell

jessebraham, 2021-11-22:
Dual-licensing MIT or Apache-2.0 is my usual go-to as far as software goes.

For the hardware side of things, I'm not sure if something else is more applicable or not. I often see Creative Commons, the CERN Open Hardware License, or the TAPR Open Hardware License as well.

With that said, my understanding is that we can also just license the hardware under MIT or Apache-2.0 as well, so for simplicity's sake that is probably the best choice, IMO.

0Grit, 2021-11-22:
@beriberikix

beriberikix, 2021-11-22:
IANAL :) I usually point people to https://choosealicense.com/.

My *personal* choice is usually Apache 2 for software for the broadest adoption and only alternative licenses like MIT, BSD-3 or GPL(x) if there is dependency on existing licenced code. Dual-licencing sometimes requires adopters to bring in lawyers, but not always.

For all other assets (except hardware) I recommend cc-by-sa-4.0 and for hardware CERN OHL.

pedrominatel, 2021-11-24:
Hi everybody,

If no objections about the license, we can merge (#8) and close this issue.

closed 2021-11-25 by jessebraham