esp-rs/esp-rust-board #31: Relicense under a permissive, non-copyleft license

TheButlah, 2022-02-20:
CERN-OHL-S v2 is a copyleft, non-permissive license, much in the same way that GPL is. For the same motivations on why the rest of the rust ecosystem is licensed under Apache 2.0/MIT, could this project please re license under a new license that is permissive and not reciprocal/viral? 

I don't know much about hardware licenses, but according to #5 it seems that the particular variant of the CERN license was chosen somewhat arbitrarily. Unlike the S variant, the P variant (CERN-OHL-P v2) is permissive, which means the following (IANAL):

* Unlike S variant, modified source does not need to be licensed under the same license.
* All the license notices still need to be provided (including the notice printed on the PCB that it is a CERN-OHL-P v2 license), but unlike S variant, notices that are no longer relevant due to modification and subsequent re-license can be omitted.
* Unlike the S variant, distributing source along with the full product is merely optional.

TheButlah, 2022-02-20:
Citations: 
* [P variant user guide](https://ohwr.org/project/cernohl/wikis/uploads/8a6b5d01f71c207c49493e4d114d61e6/cern_ohl_p_v2_howto.pdf) [
* [S variant user guide](https://ohwr.org/project/cernohl/wikis/uploads/cf37727497ca2b5295a7ab83a40fcf5a/cern_ohl_s_v2_user_guide.pdf)
* https://ohwr.org/project/cernohl/wikis/Documents/CERN-OHL-version-2

TheButlah, 2022-02-20:
To clarify: the current license requires source to be provided (things can not be closed source), and it requires derivative works to use the same license (license is viral).

These are properties of licenses the rust ecosystem generally desires to avoid

georgik, 2022-02-22:
Channing license to more permissive makes sense. The board is for community and businesses, it does not make sense to force any restrictions on it.

closed 2022-02-28 by pedrominatel