esp-rs/esp-rust-board #33: SHELL_GND should be floating

TheButlah, 2022-03-01:
Disclaimer: I have almost no electronics experience. Please verify my advice with someone with actual electrical engineering experience.

I believe the `SHELL_GND` pins on the USB-C connector should not be grounded, but rather should be floating. This is to prevent ESD and EMI (?)

Citation:
https://forum.sparkfun.com/viewtopic.php?p=102538&sid=b578afcca223f70d7e8433f18a3f1d76#p102538

julian19072001, 2022-03-01:
Not exactly what I ment. You should connect it to ground through a 1 mega ohm resistor. And not disconnect it. This is both to prevent it acting as an antenna and a esd safety thing.

TheButlah, 2022-03-14:
More conflicting advice:
https://forum.allaboutcircuits.com/threads/usb-device-cable-shield-connection-grounding-it-or-not.58811/

pedrominatel, 2022-03-14:
Hi @TheButlah,

Thanks for your contribution! We just received the first testing batch and maybe more changes will be done for the second/final version.

Usually, Espressif development board connects the USB shield to GND directly. For development boards this is ok and in most cases, EMI is not relevant.
We did connect the shield to GND but this does not mean that every project should be like.

Anyway, we can add a 0R resistor to the design and the user can evaluate if it's necessary to add a capacitor or resistor ;)

TheButlah, 2022-03-14:
Sounds good! In my design I opted for a ~100K resistor, but yeah I think this is not necessary at all

closed 2022-04-12 by pedrominatel