esp-rs/esp-rust-board #34: Update missing 3d models to use ${KICAD6_3DMODEL_DIR} where possible

TheButlah, 2022-03-01:
Most of the 3d models use the old ${KISYS3DMOD} environment variable, causing all the 3d models for the basic components to be missing. KICAD 6 introduces ${KICAD6_3DMODEL_DIR} to point to the new v6 models. 

Not all the models from v5 exist in v6 - notably, I'm seeing that `SW1`, `SW2`, `Y1`, `L1` are broken - not sure if they exist in v6 or not, but at least some of them don't.

I'll submit a PR to upgrade the models to the v6 variable where I can, and leave the old variable for the missing ones. I think this will be a manual process unfortunately, but if anyone knows an automated way to do this, lmk

TheButlah, 2022-03-01:
According to [this](https://gitlab.com/kicad/code/kicad/-/issues/10459), it looks like what the Kicad maintainers have intentionally chosen is to bring over the footprints even if the 3d models are missing, and encode the footprints as having the new environment variable instead of the old one. When the main kicad 3d library eventually adds the models back in, the footprints will naturally use the new updated models.

So I guess the migration path for now is either to accept that some of the models don't exist in v6 and wait for them to be added/open an issue in kicad. I guess the alternative would be to keep the old environment variable in for the missing ones?

I think for now since I'm not sure what y'all want, I will stick with my original plan and migrate only the footprints that have v6 models.

TheButlah, 2022-03-01:
After accidentally erasing all of the alternative 3d models y'all so carefully put in there when I nuked the project with "tools -> update footprints from library -> update/reset 3d models", I realized that the "missing" v5 parts had already been imported into the 3d models library in this repo, and set as alternate models. 

I went back and manually added back in these alternate models, along with the rotation/scaling edits that were there originally. I then went ahead and switched the path variable to the new `${KICAD6_3DMODEL_DIR}` instead of the old one, as that way when those symbols eventually do get added to upstream kicad, they will supersede the local versions.

pedrominatel, 2022-03-14:
Thanks again for your contribution!

Right now I'm doing the electrical tests on the first batch and soon I'll take care of all changes, including yours.

closed 2022-04-07 by pedrominatel