esp-rs/esp-rust-board #40: The question about MCP73831-2-OT

YuaXan, 2022-04-22:
Hello, @pedrominatel. 
The BOM sheet has a charge management controllers name's MCP73831-2-OT, but I review an MCP73831/2 datasheet and found many models.
[image]

I don't know if I should choose what model.

please provide a model name, thanks.

pedrominatel, 2022-04-22:
Hi, the one used on our first batch was the **MCP73831T-2ACI/OT**, but I've also tested with the MCP73831T-2DCI/OT.

I'll update the BOM to include the full MPN **MCP73831T-2ACI/OT** as the default.

Thanks!

YuaXan, 2022-04-22:
> Hi, the one used on our first batch was the **MCP73831T-2ACI/OT**, but I've also tested with the MCP73831T-2DCI/OT.
> 
> I'll update the BOM to include the full MPN **MCP73831T-2ACI/OT** as the default.
> 
> Thanks!

I used  **MCP73831T-2ATI/OT** happened interruption of the **STAT** port with **VBAT** port.
And  **VBAT** port connects with MOSFET's drain.
When 5V connects to the broad that **VBAT** appears waveforms affect MOSFET's drain which is the cause of 3.3V precarious.
I am not sure the 3.3v precarious what the cause used  **MCP73831T-2ATI/OT**, and I will try to use **MCP73831T-2ACI/OT**  to test.

pedrominatel, 2022-04-22:
Can you share more details about your test and the results?

YuaXan, 2022-04-22:
> Can you share more details about your test and the results?

of course.
I flash reprogramming from [ESP32-C3 AT Release v2.3.0.0](https://docs.espressif.com/projects/esp-at/en/release-v2.3.0.0_esp32c3/).
When I used **MCP73831T-2ATI/OT**, Windows and Linux can't Identify the device cause of 5v to 3.3v precarious, because I try only using 3.3v and TTL to Test which is normal, but only using a USB power supply.
And then I found the same frequency of the **STAT** port as the **VBAT** port in the oscillograph, so I think **MCP73831T-2ATI/OT** is Incompatible with the broad.
Waiting A few days later for **MCP73831T-2ACI/OT** to come, I'll tell you about the new test results.

YuaXan, 2022-04-22:
> Can you share more details about your test and the results?
I dismantle the **MCP73831T-2ATI/OT** and jump wire.
[image]

pedrominatel, 2022-04-22:
Here are some measurements for both MCP options.
[image]
[image]

pedrominatel, 2022-05-02:
Any news about your issue @YuaXan?

YuaXan, 2022-05-02:
> Any news about your issue @YuaXan?
MCP73831T-2ATI/OT 
![e00d10e262c3dc2d195f6eda9391dea](https://user-images.githubusercontent.com/28106851/166223406-f7fab568-2a68-4720-90fc-98d
[image]
6f0bb7b33.png)

pedrominatel, 2022-11-07:
Hi, any updates about this issue?

Can we close this issue?

pedrominatel, 2022-11-21:
I'm closing this issue due to inactivity.

closed 2022-11-21 by pedrominatel