esp-rs/esp-rust-board #45: QR code link to the training

jonathanpallant, 2022-05-23:
Do you think you'd have space on the board for a QR code which links to the training material and/or other docs for the board?

jonathanpallant, 2022-05-23:
Controversial thought perhaps, but seeing as the module on the top-side already has the Espressif logo on it, you could shrink the Espressif logo on the bottom-side to make room for a QR code.

pedrominatel, 2022-05-25:
Hi @jonathanpallant!

Thanks for your suggestion!

I've some consideration regarding this:

- Boards are in production, we can't change anything now
- The silkscreen quality is not the best for small QR codes (considering the space we have)
- As an alternative, we can reduce the logo as you suggested and add a short link, not QR (similar to the GH link)
- QR as footprint with ENIG finishing
- Create a custom cardboard box with the links, QR code, logo, etc.

[image]

jonathanpallant, 2022-05-26:
Putting the QR code or some other link on the box would be great. How were you planning on shipping/distributing the esp-rust boards?

pedrominatel, 2022-05-27:
The same way Espressif does for any dev board. via distributors and directly for bulk.

jonathanpallant, 2022-05-27:
Ah, I meant in an ESD bag or in a box like the ESP8684-DevKitM-1 pictured above?

pedrominatel, 2022-05-31:
Sorry for the delay! I will depend. Both ways are possible.

closed 2023-02-23 by pedrominatel