esp-rs/esp-rust-board #55: Should  GPIO2 have a pull-up?

TheButlah, 2022-11-06:
In the reference manual for the ESP32-C3, the strapping pins are as follows:
[image]

It looks like GPIO2 should be pulled up, but this board appears to not do so. Is this intentional?

pedrominatel, 2022-11-07:
Hi @TheButlah

Considering this is a development board, the GPIO8 is in high impedance and the user should care about it. In a product, the recommendation is to pull-up or pull-down (depending on the pin function) any pin that is in input enabled state and high impedance during the boot.

In this particular case, it's ok not having the pull-up.

pedrominatel, 2022-11-07:
Hi,

This project was based on the [ESP32-C3-DevKitM](https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf).

The GPIO2 is floating on the DevKitM and on the Rust Board is connected to the RGB LED.

Are you facing some unexpected behavior when using the GPIO2 pulled down during the boot?

TheButlah, 2022-11-07:
I have [my](https://github.com/TheButlah/slimevr_pcb/tree/main/hardware/ferrous_slime) own PCB that I used the rust board as a reference for. I'm having USB problems (present in my v0.3.0 but not v0.1.0 strangely, and yet no modifications to the USB part of the circuit were made), and I thought that maybe the GPIO2 was the cause.

Hence the question - I see in section 6 of the esp32-c3-mini reference that they connect GPIO2 to a pull-up, so I thought that maybe my v0.1.0 and the rust board both got lucky, but my v0.3.0 was unlucky.

Is it true that a floating pin is fine?

pedrominatel, 2022-11-07:
Please take a look at page 21 in the datasheet.

[image]

Depending on the state of the GPIOs, you can face an issue during the boot process.

Can you confirm the initial state of the following GPIOs?

- GPIO9
- GPIO8
- GPIO3
- GPIO2